nicolai-wachenschwan/arc-overhang-prusaslicer-integration #2: Script won't run in prusaslicer postprocessing

iterhating, 2023-03-02:
Thanks for the quick response time. I updated to 2.5.0 (hadn't realized this computer was lagging) and that previous gcode issue is sorted, but a new one arose which **required** the activation of "Detect Bridging Perimeters" before the code would run. I don't see this anywhere in the readme so I'm not sure if it's papering over another error or was just always required and not made explicit.

Upshot of it is, the script now works when run in terminal, and I confirmed the printing path is correct in gcode viewer. However, it won't run when called by prusaslicer, and the error message is 

Failed starting the script 
C:\Users\iterhating\Documents\Octolab\FDM\arc-overhang-prusaslicer-integration\prusa_slicer_post_processing_script.py C:\Users\iterhating\AppData\Local\Temp\.17896.gcode.pp, Win32 error: 193

Any suggestions?

osvobo, 2023-03-02:
Hi, did you add both paths in your postprocessing output options? I was getting the same error when my python wasn't correctly defined. 
Here are my settings:
[image]

iterhating, 2023-03-03:
And here I just wasn't following the instructions, thanks! That fixed it.

closed 2023-03-03 by iterhating