nicolai-wachenschwan/arc-overhang-prusaslicer-integration #3: Script doesn't work when perimeter_extrusion_width is define as percent

5axes, 2023-03-02:
If the parameter perimeter_extrusion_width  is define as percentage and not as millimeter the code gCodeSettingDict.get("perimeter_extrusion_width")  doesn't return the right value.

[image]

nicolai-wachenschwan, 2023-03-03:
Do you see any kind of Flag/unique property in the gcode, e.g. "%" in that line, that would allow to decide whether it is percent?

5axes, 2023-03-03:
Yes It's not a big issue to solved .. You can just test if % is in the parameter  : 
Ie in the Gcode :
`; external_perimeter_extrusion_width = 105%`

But in this case you need to recalculate the value by getting the nozzle_diameter  and calculating the value. 

`; nozzle_diameter = 0.4`

5axes, 2023-03-03:
My Modification 

`########## Parameters  - adjust values here as needed ##########
def makeFullSettingDict(gCodeSettingDict:dict) -> dict: 
    """Merge Two Dictionarys and set some keys/values explicitly"""
    #the slicer-settings will be imported from GCode. But some are Arc-specific and need to be adapted by you.
    _perimeter_extrusion_width = 0.4
    if "%" in gCodeSettingDict.get("perimeter_extrusion_width") :
         _perimeter_extrusion_width= gCodeSettingDict.get("nozzle_diameter")* (float(gCodeSettingDict.get("perimeter_extrusion_width").strip("%"))/100)
    else :
        _perimeter_extrusion_width=float(gCodeSettingDict.get("perimeter_extrusion_width"))
    AddManualSettingsDict={
        #adapt these settings as needed for your specific geometry/printer:
        "AllowedSpaceForArcs": Polygon([[0,0],[500,0],[500,500],[0,500]]),#have control in which areas Arcs shall be generated
        "ArcCenterOffset":2, # Unit:mm, prevents very small Arcs by hiding the center in not printed section. Make 0 to get into tricky spots with smaller arcs.
        "ArcMinPrintSpeed":0.5*60,#Unit:mm/min
        "ArcPrintSpeed":1.5*60, #Unit:mm/min
        "ArcTravelFeedRate":30*60, # slower travel speed, Unit:mm/min
        "ExtendIntoPerimeter":2.0*_perimeter_extrusion_width, #min=0.5extrusionwidth!, extends the Area for arc generation, put higher to go through small passages. Unit:mm
        "MaxDistanceFromPerimeter":2*_perimeter_extrusion_width,#Control how much bumpiness you allow between arcs and perimeter. lower will follow perimeter better, but create a lot of very small arcs. Should be more that 1 Arcwidth! Unit:mm
`

nicolai-wachenschwan, 2023-03-06:
Thank you for pointing out this problem and solving it! 
Functionality is added now :)

closed 2023-03-06 by nicolai-wachenschwan