nicolai-wachenschwan/arc-overhang-prusaslicer-integration #5: Both methods doesn't work for me. Any idea why?

18Markus1984, 2023-03-03:
Hello. I now tried both methods and both doesn't work for me. I have a Windows 11 x64 system. I didn't change the code, maybe there is something that I'm missing. I would be really happy to find a solution. Likewise, I want to try to integrate the arc-support in my workflow. I'm working in a makerspace and the arc overhangs could be a way to drastically reduce the waste material.

**Methode A:**
[image]
**Methode B:**
[image]
My PrusaSlicer integration
[image]

18Markus1984, 2023-03-04:
Yes tried it but somehow after installing the newest version of python 3.11.2 it doesn't get the correct link to the new version and I really don't know how to fix this. I'm sorry for being so helpless. I'm not really into the windows terminal. I more like the linux console. 
[image]

nicolai-wachenschwan, 2023-03-04:
Hey Markus,
The Problem is with the „type hints“. You can delete those, Since they are not needed, but for debugging. Just delete „ | None“ (Same for the Otter occurences) and it should work. 
will update the Code next week as a permanent fix.

18Markus1984, 2023-03-04:
I'm feeling so stupid right now. Sorry bothering you again. I removed the | None with search and exchange. But sadly I still got this two errors in Line 353 and 415.

[image]

nicolai-wachenschwan, 2023-03-04:
No Problem, modifying Code is not an obvious solution :)
You can delete everything between „->“ and „:“ including „->“ in the lines that causes errors, until it works or gives a different error. The line Thema should end with „):“ The error are just type hints, they are not needed for the execution of the Code. I will Fix and Update the Code next week to avoid the Problem for future users:)

nicolai-wachenschwan, 2023-03-06:
Hey Markus,
Firstly: Thank you for submitting the Issue, so the code can be truely compatible to the specs in the readme :)

I removed the trouble causing type hints. It would be perfect, if you could test the updated code in your usual Python env!
I added some new functionality to reduce warping, so please make sure you also pip install "numpy-hilbert-curve" :P

@wetwool : thank you for pointing that out, since I work in Python 3.10 I totally missed that issue!

18Markus1984, 2023-03-07:
I used the new script and installed numpy-hilbert curve and used 3.8, but it doesn't work 
[image]

18Markus1984, 2023-03-07:
I worked I installed Python 3.10. Only thing you should add is to use "<PATH-TO-FILE>" between paths that have " "
[image]

The only was a little bit overextrusion add the overhang spot:
[image]
[image]

nicolai-wachenschwan, 2023-03-07:
Thank you for testing the code, I got rid of the tuple type hint, would you give it another shot?

I am happy you got it running :)

I never saw this blobs in my models before, can you post your gcode here?
The overall surface seems to have a lot of imperfections, is this usual for your printer or just the lighting of the photo?

18Markus1984, 2023-03-07:
The surface looks wired because of Chromatic aberration, this is a photo effect and has nothing to do with the model. Sorry for confusing you. The model surface finish besides the blob in the middle looks good. I can upload the gcode tommorow from my work pc.
[image]

18Markus1984, 2023-03-08:
We tried a print that would decently fail to test out how much we can go with it. That's petg. What we thought could be useful is a z-hope bevor moving to a different location because we noticed the surface got really pushed.
[image]
[image]

nicolai-wachenschwan, 2023-03-08:
PLA is already tricky to print in Arcs, but PETG will be much harder, bc of stringing, but also because mechanical strength might depend on the cooling settings. Zhop is currently not supported, as described in limitations section. + it would add to the stringing.

You will need to fiddle a lot with the settings to get it working with petg, but feel free to give it a try!

closed 2023-03-11 by nicolai-wachenschwan