nicolai-wachenschwan/arc-overhang-prusaslicer-integration #10: :1055: After Layer Change missing keyword,

Mrfibble00, 2023-03-07:
Hey, I'm new here and don't know the etiquette so I hope this is the right place to ask.
Trying to have a go at this and trying to use it through the console, I've gotten past numerous hurdles but I'm stuck here 
[image]
I shortened the script filename just to make life simpler.

In PrusaSlicer under Printer Settings the "After layer change G-code" box is empty, should I have something in there?
Thanks.

nicolai-wachenschwan, 2023-03-07:
Hey,
yes, in the standard- profiles that box is populated with the keyword ";AFTER_LAYER_CHANGE" and some other stuff. The script needs this keyword for detecting layer start.
You can fix the Issue by simply adding
;AFTER_LAYER_CHANGE 
into that box. 
Please let me know if the fix was successful :)

Mrfibble00, 2023-03-07:
Thank you, It put me on the right track, I also had to add G92 E0 to the same box. It now gives me this.
[image]
But the output gcode preview doesn't appear to have changed at in anyway at the layers you would expect it to, however the gcode has grown by 59KB.

I dare say it's something I've done incorrectly.

nicolai-wachenschwan, 2023-03-08:
Can you Post your Gcode file ? Ideally before and after running the script :)

msticninja, 2023-03-09:
Same issue here, added the layer change keyword, but it's not seeming to modify the gcode.
[Before-Script.log](https://github.com/nicolai-wachenschwan/arc-overhang-prusaslicer-integration/files/10926892/Before-Script.log)
[After-Script.log](https://github.com/nicolai-wachenschwan/arc-overhang-prusaslicer-integration/files/10926894/After-Script.log)
Extensions changed to .log to avoid Github limits.

nicolai-wachenschwan, 2023-03-09:
Thank you for the posting!
Ok, that "bug" is a funny one:
With a setting you can block certain areas in x&y. This was ment as a manual primitive filter, before more specific metrics where found.
By default the programm allowed generation of the arcs only within the space [0,500] in x&y. Since your printer has the centerpoint in the middle, the generated polys where not inside allowed space and therefore considered invalid.

I fixed it by adding the option to remove the filter + now the x&y filter is now off by default.

Please let me know, if the problem is fixed with this :)

msticninja, 2023-03-10:
Running the latest version of the script with checkForAllowedSpace set to false still gives the same results, unfortunately.

nicolai-wachenschwan, 2023-03-10:
There was an additional setting: you had all the extrusion-widths to 0. I dont know why this was configured this way, but I would recommend to specify the extrusion width of perimeter, external perimeter and solid infill explicite in prusa slicer.
Therefore I added a userwarning.
I testet your file with all extrusion widths set to 0.45, it works then.
Please note: If you want to print arcs in very small spaces you need to change the parameters accordingly. e.g. higher extendIntoPerimeter, maybe deactivate useLeastCenterpoints,
If the hilbert infill (pastet instead of solid infill above arcs) is not displayed correctly you might reduce the minimum number of points in the  hilbert infill function. By default it prints only curve parts beeing at least 5 points long.

closed 2023-03-14 by nicolai-wachenschwan