nicolai-wachenschwan/arc-overhang-prusaslicer-integration #9: Cura and arc-overhang script

Almazick, 2023-03-07:
I just tried your script and I'm getting an error using Cura.  I don't know if I'm doing something wrong or not.  

`root@Debian:/mnt/usb# python3 ./arc-overhang.py ./test.gcode
Traceback (most recent call last):
  File "/mnt/usb/./arc-overhang.py", line 1156, in <module>
    main(gCodeFileStream,path2GCode)
  File "/mnt/usb/./arc-overhang.py", line 114, in main
    parameters=makeFullSettingDict(gCodeSettingDict)
  File "/mnt/usb/./arc-overhang.py", line 64, in makeFullSettingDict
    "ExtendIntoPerimeter":1.0*gCodeSettingDict.get("perimeter_extrusion_width"), #min=0.5extrusionwidth!, extends the Area for arc generation, put higher to go through small passages. Unit:mm
TypeError: unsupported operand type(s) for *: 'float' and 'NoneType'
root@Debian:/mnt/usb# python3 ./arc-overhang.py ./test.gcode
Traceback (most recent call last):
  File "/mnt/usb/./arc-overhang.py", line 1156, in <module>
    main(gCodeFileStream,path2GCode)
  File "/mnt/usb/./arc-overhang.py", line 114, in main
    parameters=makeFullSettingDict(gCodeSettingDict)
  File "/mnt/usb/./arc-overhang.py", line 64, in makeFullSettingDict
    "ExtendIntoPerimeter":1.0*gCodeSettingDict.get("perimeter_extrusion_width"), #min=0.5extrusionwidth!, extends the Area for arc generation, put higher to go through small passages. Unit:mm
TypeError: unsupported operand type(s) for *: 'float' and 'NoneType'
`

stmcculloch, 2023-03-07:
This script is currently only compatible with gcode created by PrusaSlicer

nicolai-wachenschwan, 2023-03-07:
PrusaSlicer puts the used config at the end of the file and uses special keywords in the gcode, the script listens for.
Porting the script to cura requires changing all this specific code, you can try this if you want.

closed 2023-03-07 by nicolai-wachenschwan

5axes, 2023-03-08:
Poeting this script to Cura is not so easy. Reading the Cura parameter  is not a big issue. But as Cura doesn't tag the Overhang code as Prusa do it will be difficult to analyse and separate the code to convert the overhang areas in circular interpolation.