nicolai-wachenschwan/arc-overhang-prusaslicer-integration #11: Can't use the script

rox97, 2023-03-10:
Hi. I tried to use the script, both in prusaslicer and via console.
In prusaslicer I got this error: 
`Post-processing script /us/local/Cellar/python@3.10/3.10.10_1/bin/python3.10 /Users/andrearossetti/Library/Application)
Support/PrusaSlicer-alpha/prusa_slicer_post_processing_script.py on file /var/folders/7q/qzjmm0d121_gtxOycd6vryr0000gn/T/.40741.gcode.pp failed.
Error code: 1
Output:
Traceback (most recent call last):
File "/Users/andrearossetti/Library/Application Support/PrusaSlicer-alpha/prusa_slicer_post_processing_script.py", line 1158, in
<module>
main(gCodeFileStream,path2GCode)
File "/Users/andrearossetti/Library/Application Support/PrusaSlicer-alpha/prusa_slicer_post_processing _script.py", line 374, in
main
input("Push enter to close this window")
EOFError: EOF when reading a line`

And via console I can finish the script but it doesn't change anything. This is what I got from console:
`/usr/local/Cellar/python@3.10/3.10.10_1/bin/python3.10 /Users/andrearossetti/Library/Application\ Support/PrusaSlicer-alpha/prusa_slicer_post_processing_script.py /Users/andrearossetti/Downloads/Unito_0.2mm_PETG_MK3S_1h15m.gcode
/Users/andrearossetti/Library/Application Support/PrusaSlicer-alpha/prusa_slicer_post_processing_script.py:1070: Travel Moves may cross the outline and therefore cause artefacts in arc generation.
last read linenumber: 93755
layers: 247
/Users/andrearossetti/Library/Application Support/PrusaSlicer-alpha/prusa_slicer_post_processing_script.py:492: Layer 0: no height found, using layerheight default!
write to file
Code executed successful
Push enter to close this window`

nicolai-wachenschwan, 2023-03-10:
Is a " " (empty space) in the path to your python exe or the script?

rox97, 2023-03-10:
Yes
`/usr/local/Cellar/python@3.10/3.10.10_1/bin/python3.10 /Users/andrearossetti/Library/Application\ Support/PrusaSlicer-alpha/prusa_slicer_post_processing_script.py;`

nicolai-wachenschwan, 2023-03-10:
You need to mask the empty spaces as described here: https://manual.slic3r.org/advanced/post-processing
Also the path needs to be the full path like C:\....

closed 2023-03-10 by nicolai-wachenschwan

rox97, 2023-03-10:
I'm using macOS and the path is the full path. The path in macOS doesn't start with C:... Also I've tried to change the path to the script with a path without a space and it gives the same error. Trying with the ! mask you sent  me, the error is that he can't find the file because the path is incomplete. Could it be that this script works only on windows?
This is the the paths with the mask:
`/usr/local/Cellar/python@3.10/3.10.10_1/bin/python3.10 /Users/andrearossetti/Library/Application! Support/PrusaSlicer-alpha/prusa_slicer_post_processing_script.py;`
and the is the error:
`Post-processing script /us/local/Cellar/python@3.10/3.10.10_1/bin/python3.10 /Users/andrearossetti/Library/Application!
Support/PrusaSlicer-alpha/prusa_slicer_post_processing_script.py on file /var/folders/7q/qzimm0d121_gtxOycd6vryr0000gn/T/.55218.gcode.pp failed.
Error code: 2
Output:
usr/local/Cellar/python@3.10/3.10.10_1/bin/python3.10: can't open file '/Users/andrearossetti/Library/Application!': [Errno 2] No such file or directory`

rox97, 2023-03-10:
`/usr/local/Cellar/python@3.10/3.10.10_1/bin/python3.10 /Users/andrearossetti/Downloads/arc-overhang-prusaslicer-integration-main/prusa_slicer_post_processing_script.py;`
This is the other path that I tried to use. As you can see, there are no spaces but the error it's the same: `EOFError: EOF when reading a line`

nicolai-wachenschwan, 2023-03-11:
Probably, but I cant fix that Issue, as i dont have a mac to test solutions. But least the console works.

PR #12 (2023-03-12): Fix MacOS PrusaSlicer bug
Deleting the line to close the program and letting it close itself automatically, i fixed the EOF error that occurred in PrusaSlicer on MacOS. Fix #11 issue
Some other instructions to use white spaces in paths both on unix like systems and windows