nicolai-wachenschwan/arc-overhang-prusaslicer-integration #14: TypeError in Python 3.10.6

stingwraith, 2023-03-13:
When running the script on a gcode file I'm getting the following error:

`python prusa* 16x*
Traceback (most recent call last):
  File "/home/username/Documents/arc-overhang-prusaslicer-integration/prusa_slicer_post_processing_script.py", line 1173, in <module>
    main(gCodeFileStream,path2GCode, skipInput)
  File "/home/username/Documents/arc-overhang-prusaslicer-integration/prusa_slicer_post_processing_script.py", line 109, in main
    parameters=makeFullSettingDict(gCodeSettingDict)
  File "/home/username/Documents/arc-overhang-prusaslicer-integration/prusa_slicer_post_processing_script.py", line 58, in makeFullSettingDict
    "ExtendIntoPerimeter":1.5*gCodeSettingDict.get("perimeter_extrusion_width"), #min=0.5extrusionwidth!, extends the Area for arc generation, put higher to go through small passages. Unit:mm
TypeError: unsupported operand type(s) for *: 'float' and 'NoneType'`

Any help in solving this would be very much appreciated. I'm excited to give arc supports a try!

nicolai-wachenschwan, 2023-03-13:
Hey,
The Error arrises, because it was unable to read the parameter "external_perimeter_extrusion_width" from the gcode.
What you can do about it: 
1. Check the gcode file. At the very bottom there should be the configuration. Search for the parameter and make sure it is a numeric value, e.g. 0.45 (no comma, no unit!)
2. try a prusa-default print-setting
3. If nothing helps: fix it on the surface by adding the parameter in the paramters section manually. If you do so you need to repeat this for other parameters portet from prusaslicer. The error notice gives you a hint which one it is.
What Operating-System do you use?

Let me know, if anything succedes!

closed 2023-03-13 by stingwraith

nicolai-wachenschwan, 2023-03-13:
Happy to help! Usually the script tells you which one Right above. Also There are only 2 cases : User relative e distances and detect bridges. Example: no brigde detection->enable Bridge detection. 
Othello Guidelines for settings in readme.