nicolai-wachenschwan/arc-overhang-prusaslicer-integration #16: Two errors when trying to run on Vase mode PrusaSlicer 2.6.0-alpha5

salamanders, 2023-03-15:
Setup was easy:

```bash
pip3 install shapely numpy matplotlib numpy-hilbert-curve
```

But when trying to modify a vase-mode print, I got:

```
$ python3 prusa_slicer_post_processing_script.py egg3_arc.gcode 
unmatching setting: [' end_gcode ', '{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+1, max_print_height)} F720 ; Move print head up{endif}\\nG1 X0 Y200 F3600 ; park\\n{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+49, max_print_height)} F720 ; Move print head further up{endif}\\nG4 ; wait\\nM221 S100 ; reset flow\\nM900 K0 ; reset LA\\n{if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3|@0.25 nozzle MK3).*/}M907 E538 ; reset extruder motor current{endif}\\nM104 S0 ; turn off temperature\\nM140 S0 ; turn off heatbed\\nM107 ; turn off fan\\nM84 ; disable motors\\n; max_layer_z ', '[max_layer_z]']

prusa_slicer_post_processing_script.py:1074: Overhang detection disabled. Activate for script success!
prusa_slicer_post_processing_script.py:111: Incompatible Settings used!

Can not run script, gcode unmodified. Press enter to close.
```

So two quick questions:

1. How do I enable overhang detection? (PrusaSlicer, I assume the Alpha does it the same as the released version)
2. That `unmatching setting` part, not sure what the cause is.

nicolai-wachenschwan, 2023-03-16:
The script relays on comments PrusaSlicer generates in the GCode. only if PrusaSlicer detects Overhangs and therefore generates BridgeInfill the script will do its work by converting BridgeInfill into Arcs.
The Error is raised, because the overhang detection in PrusaSlicer is mandatory for the script to work.

Also:
Vase-Mode does not generate Infill=>Script does not generate Arcs.

I clarified the warnings. For easier understanding.

nicolai-wachenschwan, 2023-03-16:
Regarding Question 1: Print settings tab->Layers and ...->overhang detection (if not visible set your level to expert to display all settings)

closed 2023-03-16 by nicolai-wachenschwan

salamanders, 2023-03-26:
Huh, not seeing "Overhang detection" maybe it changed in 2.6.0 alpha 5.
[image]

nicolai-wachenschwan, 2023-03-26:
Please Download the latest version of the script. I clarified the error message. The Parameter is called detect bridging perimeters

salamanders, 2023-03-26:
Whohoo, that was it, thank you!