nicolai-wachenschwan/arc-overhang-prusaslicer-integration #18: Overhang detection disabled in PrusaSlicer.

mehmetpakkabaggen, 2023-03-19:
Hi, when i try to export  get error that i dont htve  Overhang detection in PrusaSlicer. i cant find any settings for that neither online or in the app itself.
here is the full error message:
C:\temp\prusa_slicer_post_processing_script.py:1080: Overhang detection disabled in PrusaSlicer. Activate in PrusaSlicer for script success!
C:\temp\prusa_slicer_post_processing_script.py:111: Incompatible PursaSlicer-Settings used!
Can not run script, gcode unmodified. Press enter to close.

nicolai-wachenschwan, 2023-03-19:
It is here, you need to set your expirience to advanced or expert (top right corner) to see this.

[image]
This is german text, but the position will be the same for all languages.

closed 2023-03-19 by nicolai-wachenschwan

jdf221, 2023-03-20:
It's called "Detect bridging perimeters" in the English version seemingly:

[image]

nicolai-wachenschwan, 2023-03-20:
Perfect, thank you for your help! I clarified the warning with matching text.