nicolai-wachenschwan/arc-overhang-prusaslicer-integration #19: Error when exporting the g-code after slicing.

billfj4045, 2023-03-19:
Just went through the install process and sliced a model. Prusa Slicers seems to run the slice process... but gives me an error when I go to Export the G-code: "Post-processing script C:\Users\billf\AppData\Local\Programs\Python\Python311\python.exe C:\TEMParc\prusa_slicer_post_processing_script.py on file C:\Users\Public\Documents\Wondershare\CreatorTemp\.28732.gcode.pp failed.
Error code: 1"

What is Wondershare? and how did it get included in the script? 
I haven't been able to get any g-code out to confirm the script is actually doing anything;  the slice from prusa slicer still looks like a normal slice. 

thank you.

billfj4045, 2023-03-19:
Update: Removed "Wondershare"... not sure how what where that came from. Still get an error on export:
>>>Post-processing script C:\Users\billf\AppData\Local\Programs\Python\Python311\python.exe C:\TEMParc\prusa_slicer_post_processing_script.py on file C:\Users\billf\AppData\Local\Temp\.3140.gcode.pp failed.
Error code: 1

When I open the .3140.gcode file in notepad, I get what appears to be a normal slice file, but with no arcs for the overhangs?

billfj4045, 2023-03-19:
Update: if I use the CMD, change directory to C:\temparc and run the following... C:\temparc> C:\temparc\prusa_slicer_post_processing_script.py C:\temparc\Body1_0.8n_0.4mm_PETG_MK3S_14m.gcode

I get:
>>>>>
C:\temparc\prusa_slicer_post_processing_script.py:1083: Your Bridging Speed is set to 22 mm/s in PrusaSlicer. This can cause problems with warping.<=5mm/s is recommended
C:\temparc\prusa_slicer_post_processing_script.py:1089: PrusaSlicer-Setting: Travel Moves may cross the outline and therefore cause artefacts in arc generation.
last read linenumber: 7791
layers: 51
C:\temparc\prusa_slicer_post_processing_script.py:497: Layer 0: no height found, using layerheight default!
overhang found layer 26: 1
←[F←[Kwhile executed: 0 1
C:\Users\billf\AppData\Local\Programs\Python\Python311\Lib\site-packages\shapely\predicates.py:540: invalid value encountered in contains
modifying GCode
oldpolys found in layer: 27
making hilbert surface
modifying GCode
oldpolys found in layer: 28
making hilbert surface
modifying GCode
oldpolys found in layer: 29
modifying GCode
oldpolys found in layer: 30
modifying GCode
oldpolys found in layer: 31
modifying GCode
oldpolys found in layer: 32
modifying GCode
oldpolys found in layer: 33
modifying GCode
oldpolys found in layer: 34
modifying GCode
overwriting file
Script execution complete.
Press enter to exit.
>>>>>
The original g-code that was used is now updated to include the arcs, but didn't update the bridging speeds or "Travel Moves may cross the outline".
[image]

Now I need to get the code running with Prusa Slicer.

billfj4045, 2023-03-20:
Finally... I got the script to run directly from Prusa. I see that the script pop-up window gives clues on what setting should be adjusted. Once exported, I can go back in... make a few changes and rerun the export. While this print isn't a pro-level overhang print, it was a proof of concept. I think I see a few more setting changes that might improve this print too. You guys are awesome; how can I buy you a coffee/espresso so show my appreciation for your hard work on this?    

[image]
[image]
[image]

nicolai-wachenschwan, 2023-03-20:
I am happy you got it worked out. 
Some parameters need to be tuned to achieve very good results, as your printer might have other cooling fans than mine,etc.

I coded this as a hobby and I am happy if I push the arc-overhang project, no need for money, but thx anyway :)

closed 2023-03-20 by nicolai-wachenschwan