nicolai-wachenschwan/arc-overhang-prusaslicer-integration #21: "can not run script, gcode unmodified" error

danwiley, 2023-03-20:
trying to run the script on ender 3 v2 profile but im getting this error - "can not run script, gcode unmodified". i have noticed that when i change my profile to one of prusa printers the scripts run just fine. how can i fix this issue with my profile? or an update on the script will make creality compatible too? im so excited with this project and cant wait to try it!

nicolai-wachenschwan, 2023-03-20:
Probably PrusaSlicer Setting, please copy the Full printout from the Script and Paste it here.

danwiley, 2023-03-20:
Im sorry, printout?

nicolai-wachenschwan, 2023-03-20:
The Script opens a new window. Usually a Black one. It writes text there. I need you to copy that Text and paste it here.

danwiley, 2023-03-20:
[image]
it wont let me copy so i sent a screenshot. hops it helps

danwiley, 2023-03-20:
well i kinda feel stupid here because after having a second look, the problem was obvious. had to turn on detect bridging perimeters ;)

closed 2023-03-20 by danwiley