nicolai-wachenschwan/arc-overhang-prusaslicer-integration #22: TypeError: unsupported operand type(s) for *: 'float' and 'NoneType'

XenonFlits, 2023-03-20:
Running Prusa2.3.0

Error I get is
---
Error code: 1
Output:
Traceback (most recent call last):
File "/home/pi/Documents/3Dprinting/postprocessing/NoOverhang.py", line 1179, in <module>
main(gCodeFileStream,path2GCode, skipInput)
File "/home/pi/Documents/3Dprinting/postprocessing/NoOverhang.py", line 109, in main
parameters=makeFullSettingDict(gCodeSettingDict)
File "/home/pi/Documents/3Dprinting/postprocessing/NoOverhang.py", line 58, in makeFullSettingDict
"ExtendIntoPerimeter":1.5*gCodeSettingDict.get("perimeter_extrusion_width"), #min=0.5extrusionwidth!, extends the Area for arc generation, put higher to go through small passages. Unit:mm
TypeError: unsupported operand type(s) for *: 'float' and 'NoneType'
---
Header of prntfile is
---
; generated by PrusaSlicer 2.3.0 on 2023-03-20 at 19:39:36 UTC

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.42mm

M107
M190 S50 ; set bed temperature and wait for it to be reached
M104 S220 ; set temperature
;TYPE:Custom
START_PRINT
M109 S220 ; set temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for etc etc
----
What is the problem? Is it a bug? Or do I have to correct a Prusa setting?

nicolai-wachenschwan, 2023-03-21:
Thank you @maruhe, could not tell it better :)

maruhe, 2023-03-21:
@nicolai-wachenschwan we thank you, have been waiting for long time ;-)

if you may integrate to also read  ; superslicer_config = begin pls also add the end of settings - if other plugins add Text at the end there comes a lot of output from your script

    for line in gcodeLines:
        if "; prusaslicer_config = begin" in line:
            isSetting=True
            continue
        if "; prusaslicer_config = end" in line:
            isSetting=False
            continue

nicolai-wachenschwan, 2023-03-24:
updated the code, would be happy, if you could test it with superslicer

closed 2023-03-24 by nicolai-wachenschwan