nicolai-wachenschwan/arc-overhang-prusaslicer-integration #24: hilbert surface gets printed again on top surface

maruhe, 2023-03-22:
attached the models I created to test ArcOverhangs

may it's because the thickness of the overhang is quite low
I reduced BottomLayers to 3 and after ArcOverhangs and hilbert surface the infill starts (even printing slow)
but after the infill it starts again with hilbert surface printing super slow and this destried the prints (I think because the hot nozzle was nearby / touching the overhang-area for long time

I'm printing ABS and the ArcOverhangs work great!
just the following layers need to get printed faster
is there an option to limit the number of hilbert layers and the number of slow layers?

[arc.zip](https://github.com/nicolai-wachenschwan/arc-overhang-prusaslicer-integration/files/11035850/arc.zip)

while testing I also saw some moves in the infill above the hilbert that have been ultra slow and creating little blobs on the outer surface
I didn't found that part in the gcode but it felt like a G1 command without feedrate using an old slow feedrate

nicolai-wachenschwan, 2023-03-24:
Yes, there are 2 parameters defined to this usecase. You need to adapt the parameters in the "parameter" section of the script according to your needs.
Navigate to Special Cooling Settings. There you find specialCoolingZdist, which is the Z distance the special cooling settings are applied and therefore every solid infill within beeing convertet to hilbert. Default is 3mm, but you can lower it to exclude the top surfaces.

PrintSpeed:
change "aboveArcsPerimeterPrintSpeed" and "aboveArcsInfillPrintSpeed" to modify the printing speeds according to your needs.

closed 2023-03-24 by nicolai-wachenschwan

maruhe, 2023-03-24:
specialCoolingZdist is a nice hint, before it looked like this 
in case I will fix something I_ll create a pull request :-)
[image]

i still get this ultra-slow line but din't found it yet but I found a missing linebreak
[image]
also saw missing linebreaks in other files before:
[image]

billfj4045, 2023-03-24:
I presume that these changes are in the .py script itself and not in Prusa Slicer. That might be a step out of my league, but will make a copy of the original script and see what I can do to make those adjustments. Fingers crossed!