nicolai-wachenschwan/arc-overhang-prusaslicer-integration #26: Weird solid infill

coolxm, 2023-03-22:
dont know how to describe this
[image]

5axes, 2023-03-22:
Hilbert "Style" Infilll  : https://en.wikipedia.org/wiki/Hilbert_curve because unfortunatly it's not a Real Hilbert curve as you don't have just one start point.

nicolai-wachenschwan, 2023-03-24:
The line errors occur, because the previous tool position gets deleted, when deleting the old solid infill.
Haven't had the time to fix this yet. If you have the time at spare I would be happy if you could contribute a bug fix via pull-request. 
The problem lies in the  prepareDeletion Method, starting at line 696 (current).

closed 2023-03-24 by nicolai-wachenschwan

ghylander, 2023-05-12:
Having this issue as well, have you had he time to look into it?