nicolai-wachenschwan/arc-overhang-prusaslicer-integration #28: Problem with modules on mac

bifaciall, 2023-03-22:
when I try to run it, it says there is no module named shapely-> ergo i hadnt installed it; thing is: I have

nicolai-wachenschwan, 2023-03-24:
probably you have multiple python executables. (Kind of usual).
Fix: open terminal and insert the complete path of the python-executable, just as the first path in the prusaSlicer script.
 Than append the mac-command for pip install shapely and press enter.
 =>will install the lib in the right python environment.
 
 Please try this and report if it was successful :)

salamanders, 2023-03-26:
I was able to get it to execute on a mac with default python pip install steps.   +1 to "you may have installed the dependencies in a different place"

Henk72, 2023-03-26:
Thanks for the answer, great solution

bifaciall, 2023-03-29:
> probably you have multiple python executables. (Kind of usual).
>
> Fix: open terminal and insert the complete path of the python-executable, just as the first path in the prusaSlicer script.
>
>  Than [...]

I will try it when im home

closed 2023-04-12 by nicolai-wachenschwan