nicolai-wachenschwan/arc-overhang-prusaslicer-integration #40: Post-processing script Error

EduardoMDSousa, 2023-04-02:
Good afternoon, I am not able to understand the reason for this error,
I use another script accelerations.py, this one works fine, but when I try to use prusa_slicer_post_processing_script.py, I always get the error below.
I already tested with pyton 3.10 and 3.11
all libraries are installed

thanks

Post-processing script 
C:\python-3.10.0\scripts\prusa_slicer_post_processing_script.py 
on file C:\Users\Eduardo\AppData\Local\Temp\.4904.gcode.pp failed.
Error code: 1

EduardoMDSousa, 2023-04-04:
Post-processing script C:\Users\Eduardo\AppData\Local\Programs\Python\Python310\python.exe C:\python-3.10.0\scripts\prusa_slicer_post_processing_script.py on file C:\Users\Eduardo\AppData\Local\Temp\.7432.gcode.pp failed.
Error code: 1

amatulic, 2023-04-09:
You get better diagnostic information if you run the script from the commandline. I encountered a TypeError and filed issue #42. It may be the same as what you encountered, but there's no way to know unless you run it from the commandline.

nicolai-wachenschwan, 2023-04-12:
When executing the script there opens a new window with detailed informaion. If it closes to quick try running it from the commandline.

closed 2023-04-21 by nicolai-wachenschwan