nicolai-wachenschwan/arc-overhang-prusaslicer-integration #42: Script error

amatulic, 2023-04-08:
I get the following error when attempting to run the script on the attached gcode.
`Traceback (most recent call last):
  File "c:\standalone\prusa_slicer_post_processing_script.py", line 1182, in <module>
    main(gCodeFileStream,path2GCode, skipInput)
  File "c:\standalone\prusa_slicer_post_processing_script.py", line 109, in main
    parameters=makeFullSettingDict(gCodeSettingDict)
  File "c:\standalone\prusa_slicer_post_processing_script.py", line 76, in makeFullSettingDict
    "ArcWidth":gCodeSettingDict.get("nozzle_diameter")*0.95, #change the spacing between the arcs,should be nozzle_diameter`

I am using the latest stable release of PrusaSlicer, 2.5.2. My Python is 3.10 and I have installed the necessary libraries.

nicolai-wachenschwan, 2023-04-12:
The problem is that you defined a non numeric extrusion width in prusaslicer. Please define it like "0.45" mm or another numeric value. all extrusion widths are needed as numeric values, e.g. 110% is also accepted .

closed 2023-04-12 by nicolai-wachenschwan

amatulic, 2023-04-13:
> The problem is that you defined a non numeric extrusion width in prusaslicer. Please define it like "0.45" mm or another numeric value. all extrusion widths are needed as numeric values, e.g. 110% is also accepted .

Er, no, I did no such thing. I have never touched the extrusion width settings in PrusaSlicer, I use the defaults, and they are all set to 0.45 except for top solid infill at 0.4 and support material at 0.35. These are the PrusaSlicer defaults for the "0.20mm QUALITY @MK3" preset.

Given that they are already numeric values, what else could be the problem?