nicolai-wachenschwan/arc-overhang-prusaslicer-integration #43: Arcs aren't created in the right spot

planeguy12, 2023-04-10:
The arcs only form when bridging over infill (a top layer) but wont form over the 90 degree overhang.

nicolai-wachenschwan, 2023-04-12:
Can you post an image of the geometry and the detailed information the programm prints in its window or the console.

dertester, 2023-05-05:
Same here.
[image]
[image]

C:\Users\alexa\Desktop\prusa_slicer_post_processing_script.py:504: Layer 0: no height found, using layerheight default!
overhang found layer 51: 1 Z: 10.20
C:\Users\alexa\Desktop\prusa_slicer_post_processing_script.py:541: Layer 50: No ExternalPerimeterPolys found in prev Layer
C:\Users\alexa\Desktop\prusa_slicer_post_processing_script.py:168: Skipping Polygon because no StartLine Found
modifying GCode
oldpolys found in layer: 52
making hilbert surface
modifying GCode
oldpolys found in layer: 53
making hilbert surface
modifying GCode
oldpolys found in layer: 54
making hilbert surface
modifying GCode
oldpolys found in layer: 55
modifying GCode
oldpolys found in layer: 56
modifying GCode
oldpolys found in layer: 57
modifying GCode
oldpolys found in layer: 58
modifying GCode
oldpolys found in layer: 59
modifying GCode
oldpolys found in layer: 60
modifying GCode
oldpolys found in layer: 61
modifying GCode
oldpolys found in layer: 62
modifying GCode
oldpolys found in layer: 63
modifying GCode
oldpolys found in layer: 64
modifying GCode
oldpolys found in layer: 65
modifying GCode
oldpolys found in layer: 66
modifying GCode
oldpolys found in layer: 67
modifying GCode
overwriting file
Script execution complete.
Press enter to exit.

nicolai-wachenschwan, 2023-05-06:
This is a bug related to this geometry. The Algorithm needs some start point. Therefore it searches some non bridging external perimeter. 
your test has no perimeters, as the lower bit is to narrow and the bridging is printed in one piece. 

=>make the lower part bigger diameter and test again please

dertester, 2023-05-14:
That worked. I increased radius and added a little edge at the top.

joechorn, 2023-09-30:
I think in case of concentric geometries, the used algorithm does not work propperly / there could be better solutions. If the overhanging geometry is limited by concentric circles on the outside and on the inside (where it is connected to the model), I would propose the following algorithm:
1. set the starting point to the common center of the two concentric circles.
2a. if the overhanging geometry is a closed cicle, just start with the generation of the overhanging layer
2b. if the overhanging geometry is a circular ring, use the same startingpoint, but do not start to print the generated geometry  until the generated spiral is coincident to the actual geometry.

Maybe it could be possible to improve results in general, if the algorithm tries to set the (virtual) starting point as near to the center of the printed geometry instead of forcing it to be on a perimeter.