nicolai-wachenschwan/arc-overhang-prusaslicer-integration #46: TypeError still being encountered in script

amatulic, 2023-04-13:
OK, I am going to try this again, and address the closing comment in issue #42.

Trying to run the script on the attached gcode file (somewhat different than in #42) produces the following message:

    INFO:PrusaSlicer Setting not in the expected format, but added into script dictionary: [' end_gcode ', '{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+1, max_print_height)} F720 ; Move print head up{endif}\\nG1 X0 Y210 F7200 ; park\\n{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+49, max_print_height)} F720 ; Move print head further up{endif}\\n{if has_wipe_tower}\\nG1 E-15 F3000\\n{else}\\nG1 E2 F5000\\nG1 E2 F5500\\nG1 E2 F6000\\nG1 E-15 F5800\\nG1 E-20 F5500\\nG1 E10 F3000\\nG1 E-10 F3100\\nG1 E10 F3150\\nG1 E-10 F3250\\nG1 E10 F3300\\n{endif}\\n\\nM140 S0 ; turn off heatbed\\nM107 ; turn off fan\\n\\n; Unload filament\\nM702 C\\n\\nG4 ; wait\\nM221 S100 ; reset flow\\nM900 K0 ; reset LA\\n{if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3|SOLUBLE|@0.25 nozzle MK3).*/}M907 E538 ; reset extruder motor current{endif}\\nM104 S0 ; turn off temperature\\nM84 ; disable motors\\n; max_layer_z ', '[max_layer_z]']
    Traceback (most recent call last):
      File "c:\standalone\prusa_slicer_post_processing_script.py", line 1189, in <module>
        main(gCodeFileStream,path2GCode,path2Output,skipInput)
      File "c:\standalone\prusa_slicer_post_processing_script.py", line 109, in main
        parameters=makeFullSettingDict(gCodeSettingDict)
      File "c:\standalone\prusa_slicer_post_processing_script.py", line 76, in makeFullSettingDict
        "ArcWidth":gCodeSettingDict.get("nozzle_diameter")*0.95, #change the spacing between the arcs,should be nozzle_diameter
    TypeError: can't multiply sequence by non-int of type 'float'

The "INFO" message above appears to be from the "End G-code" setting, which is the stock setting for a Prusa MK3S+MMU2S.

The closing comment in #42 said that the extrusion width is not a number.

That is not the case here. I am using stock PrusaSlicer 2.5.2 settings for a Prusa MK3S with MMU. I'm using the "0.20mm QUALITY @MK3" preset. No changes to any parameters. The extrusion widths are all set to a number, 0.45 for all except top solid infill at 0.4 and support material at 0.35. These are all default values.

If PrusaSlicer is emitting incorrect g-code, then please identify what that is so you or I can report a bug on the PrusaSlicer github repository.

amatulic, 2023-04-14:
Still getting the same error on the gcode attached to this issue. The only difference is the line numbers:
    INFO:PrusaSlicer Setting not in the expected format, but added into script dictionary: [' end_gcode ', '{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+1, max_print_height)} F720 ; Move print head up{endif}\\nG1 X0 Y210 F7200 ; park\\n{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+49, max_print_height)} F720 ; Move print head further up{endif}\\n{if has_wipe_tower}\\nG1 E-15 F3000\\n{else}\\nG1 E2 F5000\\nG1 E2 F5500\\nG1 E2 F6000\\nG1 E-15 F5800\\nG1 E-20 F5500\\nG1 E10 F3000\\nG1 E-10 F3100\\nG1 E10 F3150\\nG1 E-10 F3250\\nG1 E10 F3300\\n{endif}\\n\\nM140 S0 ; turn off heatbed\\nM107 ; turn off fan\\n\\n; Unload filament\\nM702 C\\n\\nG4 ; wait\\nM221 S100 ; reset flow\\nM900 K0 ; reset LA\\n{if print_settings_id=~/.*(DETAIL @MK3|QUALITY @MK3|SOLUBLE|@0.25 nozzle MK3).*/}M907 E538 ; reset extruder motor current{endif}\\nM104 S0 ; turn off temperature\\nM84 ; disable motors\\n; max_layer_z ', '[max_layer_z]']
    Traceback (most recent call last):
      File "c:\standalone\prusa_slicer_post_processing_script.py", line 1194, in <module>
        main(gCodeFileStream,path2GCode, skipInput)
      File "c:\standalone\prusa_slicer_post_processing_script.py", line 114, in main
        parameters=makeFullSettingDict(gCodeSettingDict)
      File "c:\standalone\prusa_slicer_post_processing_script.py", line 79, in makeFullSettingDict
        "ArcWidth":gCodeSettingDict.get("nozzle_diameter")*0.95, #change the spacing between the arcs,should be nozzle_diameter
    TypeError: can't multiply sequence by non-int of type 'float'

nicolai-wachenschwan, 2023-04-15:
Didnt read well enough, should be fixed now. Would be happy if you could Test it on your machine :)

amatulic, 2023-04-16:
I get the same error message as before, with the same line numbers indicated. I checked the diff of your commit and that change is in the version I am running.

If you test the g-code file I attached to the initial post in this issue, does it work for you?

nicolai-wachenschwan, 2023-04-16:
ok, this was a bit thougher than expected. Apperently a lot of values are speced as a list when using MMU. Therefore I wrote the addition to convert all tuples into their first value. Also included fallback option for manual specification and warnings which tell how to set manual fallback values.

Converted your file successfully, but for your specific geometry using the recursive function by setting the parameter "UseLeastAmountOfCenterPoints" to False could print better.

nicolai-wachenschwan, 2023-04-16:
PS: If you only use one extruder you can also slice the file for the regular MK3S and chose the filament when starting the print manual :)

nicolai-wachenschwan, 2023-04-21:
Yes, monotonic would warp terriblely.

closed 2023-04-21 by nicolai-wachenschwan