nicolai-wachenschwan/arc-overhang-prusaslicer-integration #48: Printer nocks print off buildplate

1Euro7Cent, 2023-05-01:
I've ran the script and it works fine. In the print when it generates the first few arcs it always goes back to starting pos of the next arc at the same pos. At the start of every arc a small blob appears and at that move it crashed into the print.

You can see the blobs I was talking about
[image]

[image]

1Euro7Cent, 2023-05-01:
A possibility to fix that is by adding a "zHop" like system or plan the travel moves outside of already printed paths.

Or fix the blob stuff

nicolai-wachenschwan, 2023-05-04:
Just did a quick fix in the lunch break, now fixed for real. at least it works on my machine :)

1Euro7Cent, 2023-05-04:
I think you messed up some units. It is printing now and it retracts. But it retracts like 100X slower than I've set it to(I've set it to 30 mm/s)
[image]

1Euro7Cent, 2023-05-04:
It should take less than a second. But it takes like 3 seconds(0.8 mm retract)

nicolai-wachenschwan, 2023-05-04:
Jup. Fixed….

1Euro7Cent, 2023-05-04:
The blob issue is fixed but the part bend because of that wiered layer above it. (it softened the arcs)
[image]

nicolai-wachenschwan, 2023-05-04:
Perfect! Just Use a little more cooling in the special cooling settings and experiment a bit with the paramesters:)

1Euro7Cent, 2023-05-05:
Ok. That fixes my problems. Going to find some time tuning value.

closed 2023-05-05 by 1Euro7Cent