nicolai-wachenschwan/arc-overhang-prusaslicer-integration #52: ImportError: cannot import name 'Point' from 'shapely'

Mac-EvoyING, 2023-06-06:
Hi! 
Do you have any suggestions about why the import fails? I'm using python 3.11.2 and shapely 1.8.5 in Debian testing.

nicolai-wachenschwan, 2023-11-13:
Hey, sorry for the very late answer, but this is not a script associated problem. Maybe you have multiple python versions installed and the one you use for the script execution has shapely not installed. try calling the python by full path and then pip install shapely.

cmadson, 2024-01-27:
Nicolai nailed it here. I had to update brew on my Mac, then install command line tools, then update brew again and then installed shapely again via brew. It took a couple interations to make it work. Only after did I notice that I had python and python3 and they were installing to separate libraries on my computer.