nicolai-wachenschwan/arc-overhang-prusaslicer-integration #53: Error on some conversions

mikestanich, 2023-06-17:
I apologize if this isn't the correct forum.  I'm having an issue running the script, either automatically with prusa slicer or from the command line.  I have had 1 successful use, but all others have thrown errors with the message below, I do not know how to fix this, or what causes it and would appreciate assistance either with changing my settings/model or the script to get it to work.

/home/mstanich/Programming/3dModels/3dPrinter/prusa_slicer_post_processing_script.py:504: Layer 0: no height found, using layerheight default!
overhang found layer 71: 2 Z: 11.40
/home/mstanich/Programming/3dModels/3dPrinter/prusa_slicer_post_processing_script.py:529: Layer 70: Could not fetch real StartPoint.
Traceback (most recent call last):
  File "/home/mstanich/Programming/3dModels/3dPrinter/prusa_slicer_post_processing_script.py", line 1197, in <module>
    main(gCodeFileStream,path2GCode, skipInput)
  File "/home/mstanich/Programming/3dModels/3dPrinter/prusa_slicer_post_processing_script.py", line 170, in main
    startpt=getStartPtOnLS(startLineString,parameters)
  File "/home/mstanich/Programming/3dModels/3dPrinter/prusa_slicer_post_processing_script.py", line 862, in getStartPtOnLS
    if not lsidx.is_integer():
AttributeError: 'numpy.int64' object has no attribute 'is_integer'

nicolai-wachenschwan, 2023-11-13:
Hey, thank you for posting that error! Will try to rewrite the check :)

Wasupmacuz, 2024-04-22:
Hey @nicolai-wachenschwan is this issue closed? I can take a look to see if I can replicate it if not.

derlui, 2024-08-02:
exact same issue here.

Wasupmacuz, 2024-11-30:
As far as I know, using `numpy.argmax` on a single-dimensional array will **always** produce an integer, not an array like the code here is checking for. As long as I'm correct in my assertion, you shouldn't need this conditional at all!
The new `getStartPtOnLS` function will look like this:
```
def getStartPtOnLS(ls:LineString,kwargs:dict={},choseRandom:bool=False)->Point:
    if ls.geom_type=="MultiLineString" or ls.geom_type=="GeometryCollection":
        lengths=[]
        for lss in ls.geoms:
            if lss.geom_type=="LineString":
                lengths.append(lss.length)
            else:
                print("Startline Item bizzare Type of geometry:",lss.geom_type)
                lengths.append(0)
        lsidx=np.argmax(lengths)
        ls=ls.geoms[lsidx]
    if len(ls.coords)<2:
        warnings.warn("Start LineString with <2 Points invalid")
        input("Can not run script, gcode unmodified. Press Enter")
        raise ValueError("Start LineString with <2 Points invalid")
    if len(ls.coords)==2:
        return midpoint(Point(ls.coords[0]),Point(ls.coords[1]))
    scores=[]
    curLength=0
    pts=[Point(p) for p in ls.coords]
    if choseRandom:
        return random.choice(pts)
    coords=[np.array(p) for p in ls.coords]
    for idp,p in enumerate(pts):
        if idp==0 or idp==len(pts)-1:
            scores.append(0)
            continue
        curLength+=p.distance(pts[idp-1])
        relLength=curLength/ls.length
        lengthscore=1-np.abs(relLength-0.5)#Hat-function: pointscore=1 at relLength=0.5, 0 at start or end.
        v1=coords[idp]-coords[idp-1]
        v2=coords[idp+1]-coords[idp]
        if np.linalg.norm(v1)>0 and np.linalg.norm(v2)>0:#calc angle only for non-zero-vectors
            v1=v1/np.linalg.norm(v1)
            v2=v2/np.linalg.norm(v2)
            anglescore=np.abs(np.sin(np.arccos(np.clip(np.dot(v1, v2), -1.0, 1.0))))#prefer points at corners
            anglescore*=kwargs.get("CornerImportanceMultiplier",1)
            scores.append(lengthscore+anglescore)
        else:
            scores.append(lengthscore)
    maxIndex=scores.index(max(scores))
    return pts[maxIndex]
```
It's a bit shorter than before and still works for me. If I could get some more people to test this I'll add it to my fork.