nicolai-wachenschwan/arc-overhang-prusaslicer-integration #55: Error Code 1 ?

mam326, 2023-07-01:
Hi,
What is meant by "Error Code 1" ?
I get this when trying to export the g-code.

[image]

bezalel6, 2023-08-07:
im getting the same problem... found any solution?

DumahBrazorf, 2023-08-08:
Make sure to have downloaded the .py file correctly. 
I made the mistake to downloading from the main page right clicking on the link and doing save as. Wrong! Go inside and download from the raw file link.

Big-coconut, 2023-10-23:
I have this problem, but it works if I just run the command line version.

nicolai-wachenschwan, 2023-11-13:
to fix the error I need the text in the black window for detailed info, as the prusa window always looks the same for any errors :(

Big-coconut, 2023-11-14:
Is there a way to show a terminal in Prusa Slicer? If I run the script in the command line it works fine, but in Prusa Slicer it gives the error.

Wasupmacuz, 2024-04-11:
In order to see the traceback error text, you will need to run the script using command line, as defined by the python script: "Option A) open your system console and type 'python ' followed by the path to this script and the path of the gcode file. Will overwrite the file." If this is your first time running the script, you will likely need to install dependencies using pip.
Using the Prusa slicer (option B) to run the post-processing script will automatically close the command line terminal upon encountering an error. That means @Big-coconut is out of luck as far as I know. It might be that the python.exe you linked to is outdated.

Mikaoo, 2024-11-25:
Hello everybody! Please help me, what am I doing wrong? I'm a beginner and I don't understand where I went wrong.

[image]

Mikaoo, 2024-11-25:
Hello everybody! Please help me, what am I doing wrong? I'm a beginner and I don't understand where I went wrong.
MacOS. PrusaSlicer.

[image]

Mikaoo, 2024-11-25:
Hello everybody! Please help me, what am I doing wrong? I'm a beginner and I don't understand where I went wrong.
MacOS. PrusaSlicer.

[image]

[image]

Mikaoo, 2024-11-25:
если убрать кавычки то другая ошибка 

[image]