nicolai-wachenschwan/arc-overhang-prusaslicer-integration #56: Strings added between edges

n808, 2023-07-29:
I tried the script on a small holder with overhangs. It appears to works as expected but adds a few strings between edges.
[image]

I was able to remove most warnings, in particular enabling Avoid Crossing Perimeters, which I thought could be the cause of this, but it did not help. The remaining warnings are as below. 

`
prusa_slicer_post_processing_script.py:1072: PrusaSlicer Setting  end_gcode  not in the expected key/value format, but added into the settings-dictionarry
prusa_slicer_post_processing_script.py:1086: machine_max_acceleration_e was specified as tuple/list, this is normal for using multiple extruders. For all list values First values will be used. If unhappy: Add manual fallback value by searching for ADD FALLBACK in the code. And add 'Fallback_<key>:<yourValue>' into the dictionary.
last read linenumber: 38609
layers: 69
prusa_slicer_post_processing_script.py:504: Layer 0: no height found, using layerheight default!
overhang found layer 61: 2 Z: 12.40
prusa_slicer_post_processing_script.py:529: Layer 60: Could not fetch real StartPoint.
←[F←[Kwhile executed: 0 1
predicates.py:540: invalid value encountered in contains
←[F←[Kwhile executed: 0 1
modifying GCode`

dining-philosopher, 2023-08-01:
Hello! It also gives this effect for me, but in more severe way. Interestingly the gcode viewer draws it all ok, but at actual printing it slowly prints a lot of spaghetti, and also severely underextrudes some areas.

Also it does not generate arc overhangs sometimes (mostly at the small overhangs, decreasing MinArea and MinBridgeLength does not help).

[dry_spool.tar.gz](https://github.com/nicolai-wachenschwan/arc-overhang-prusaslicer-integration/files/12231559/dry_spool.tar.gz)
[image]
(i cut off some spaghetti already)
[image]
[image]
[dry_spool.tar.gz](https://github.com/nicolai-wachenschwan/arc-overhang-prusaslicer-integration/files/12231559/dry_spool.tar.gz)

Wasupmacuz, 2024-01-05:
I also get this issue. Looking at the gcode, the issue happens when a feature reaches a dead end and must travel to a new starting point.
[image]

Wasupmacuz, 2024-04-11:
I just had to chance to look at this problem again today and I fixed both of the included travel move problems stringing (yes, it was two bugs causing this same issue). I'll upload the updated code as soon as I learn how to properly contribute to a GitHub repo lol.
Before and after pics:
[image]
[image]