nicolai-wachenschwan/arc-overhang-prusaslicer-integration #57: getting error  unsupported operand type(s)

bezalel6, 2023-08-09:
File "...overhang.py", line 1197, in <module>
    main(gCodeFileStream,path2GCode, skipInput)
  File "...overhang.py", line 106, in main
    parameters=makeFullSettingDict(gCodeSettingDict)
  File "...overhang.py", line 53, in makeFullSettingDict
    "ExtendIntoPerimeter":1.5*gCodeSettingDict.get("perimeter_extrusion_width"), #min=0.5extrusionwidth!, extends the Area for arc generation, put higher to go through small passages. Unit:mm
TypeError: unsupported operand type(s) for *: 'float' and 'NoneType'

i didnt change the script at all

nicolai-wachenschwan, 2023-11-13:
Hey, sorry for the very late answer, I heard of this error, when there is no perimeter_extrusion_width defined in the gcode or multiple extruders used... can you check the gcode file? at the very bottom there should be a collection of used parameters.

FlyForABeeGuy, 2024-05-03:
Hello, 
I'm sorry to reopen this conversation, but I have the same issue. Altough the parameter is in the gcode. I tried without ArcWelder, but it doesn't change the output. I made quickly an file in Tinkercad to test, and I use an Anycubic Kobra 2 Neo (again, I am a big noob in this). Any idea how we can move past this error? 
I couldn't upload the gcode in this box, so I changed the extension to txt

[Dazzling Gogo-0.2-16m.txt](https://github.com/nicolai-wachenschwan/arc-overhang-prusaslicer-integration/files/15203782/Dazzling.Gogo-0.2-16m.txt)

Wasupmacuz, 2024-11-30:
@FlyForABeeGuy You are getting this error because the script looks for "; prusaslicer_config = begin" to start building the settings dictionary. Your gcode instead has "; SuperSlicer_config = begin". Please note that changing this does not let your gcode work with the script because it's still missing use_relative_e_distances and possibly more required settings.