nicolai-wachenschwan/arc-overhang-prusaslicer-integration #59: arcs won't generate if they're needed in the same area 2 layers in a row.

Nathan22211, 2023-10-15:
Sliced this https://www.printables.com/model/337021-low-poly-kyogre/files with prusa and found that it doesn't generate the arcs as it should 
[image]

nicolai-wachenschwan, 2023-11-13:
Hey, sorry for the very late answer! Thank you for this test case, will try to implement it :)

SamuraiNinjaGuy, 2023-12-30:
I noticed this as well and had a bit of time tinkering with the script... It seems like when you add the specialCoolingZdist layers that you prevent the process from running correctly on those layers.

I have had difficulty following the code, but a thought did occur to me for how to fix this without (I hope) a massive rewrite.

Right now it looks like it starts searching from the bottom and when it finds a bridge infill, it replaces it with an arc, and then replaces a few layers above it with special cooling layers. This makes subsequent searches for bridge infill of layers that have the special cooling layers fail to work properly (I found a work around to keep the old bridge infill by preventing the special cooling layers from being added, but this method required me to rerun the script multiple times and had significant warping obviously...).

So for my idea, rather than starting at the bottom and searching for bridge infill, why not start at the top? That way when you modify cooling layers for the area over the arcs, you would be modifying layers that already were checked for bridge infill?

The alternative, as I see it, is to store the cooling layer information and resume the normal search recursively from that point onward. merging the changes if they require an additional arc... I don't think that is easier, but I could be wrong.