nicolai-wachenschwan/arc-overhang-prusaslicer-integration #60: int/float issue

Dragonkinn02, 2023-10-23:
https://github.com/nicolai-wachenschwan/arc-overhang-prusaslicer-integration/blame/549fcd23e6e4645cb0d48cc8d6c2830f41f5c985/prusa_slicer_post_processing_script.py#L841

Python312\lib\random throws float type cast issue, resolution may be:
random.randint(1,int(1e10)))

nicolai-wachenschwan, 2023-11-13:
Thank you, will implement that!

MrBarbie, 2023-12-18:
Can confirm this works. Before it was throwing the error below
`Traceback (most recent call last):
  File "C:\Windows.old\Users\KenW\Downloads\3D Printing\ArcOverhangs\prusa_slicer_post_processing_script.py", line 840, in <module>
    class BridgeInfill():
  File "C:\Windows.old\Users\KenW\Downloads\3D Printing\ArcOverhangs\prusa_slicer_post_processing_script.py", line 841, in BridgeInfill
    def __init__(self,pts=[],id=random.randint(1,1e10)) -> None:
                                ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\KenW\AppData\Local\Programs\Python\Python312\Lib\random.py", line 336, in randint
    return self.randrange(a, b+1)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\KenW\AppData\Local\Programs\Python\Python312\Lib\random.py", line 312, in randrange
    istop = _index(stop)
            ^^^^^^^^^^^^
TypeError: 'float' object cannot be interpreted as an integer`

nicolai-wachenschwan, 2023-12-18:
took a while, but is implemented, thank you for suggesting that fix!

closed 2023-12-18 by nicolai-wachenschwan