nicolai-wachenschwan/arc-overhang-prusaslicer-integration #61: TypeError: 'float' object cannot be interpreted as an integer

MK3D2024, 2023-11-23:
i get this error while using via console 

Microsoft Windows [Version 10.0.22631.2715]
(c) Microsoft Corporation. All rights reserved.

C:\Windows\System32>python C:\ArcOverhangs\prusa_slicer_post_processing_script.py C:\ArcOverhangs\1.gcode
Traceback (most recent call last):
  File "C:\ArcOverhangs\prusa_slicer_post_processing_script.py", line 840, in <module>
    class BridgeInfill():
  File "C:\ArcOverhangs\prusa_slicer_post_processing_script.py", line 841, in BridgeInfill
    def __init__(self,pts=[],id=random.randint(1,1e10)) -> None:
                                ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Kareem Salama\AppData\Local\Programs\Python\Python312\Lib\random.py", line 336, in randint
    return self.randrange(a, b+1)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Kareem Salama\AppData\Local\Programs\Python\Python312\Lib\random.py", line 312, in randrange
    istop = _index(stop)
            ^^^^^^^^^^^^
TypeError: 'float' object cannot be interpreted as an integer

C:\Windows\System32>python C:\ArcOverhangs\prusa_slicer_post_processing_script.py C:\ArcOverhangs\1.gcode
Traceback (most recent call last):
  File "C:\ArcOverhangs\prusa_slicer_post_processing_script.py", line 840, in <module>
    class BridgeInfill():
  File "C:\ArcOverhangs\prusa_slicer_post_processing_script.py", line 841, in BridgeInfill
    def __init__(self,pts=[],id=random.randint(1,1e10)) -> None:
                                ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Kareem Salama\AppData\Local\Programs\Python\Python312\Lib\random.py", line 336, in randint
    return self.randrange(a, b+1)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Kareem Salama\AppData\Local\Programs\Python\Python312\Lib\random.py", line 312, in randrange
    istop = _index(stop)
            ^^^^^^^^^^^^
TypeError: 'float' object cannot be interpreted as an integer

and while using inside slicer i get this error

[image]

[image]
[image]

gvancamp, 2023-12-13:
I'm getting the same error.

MrBarbie, 2023-12-18:
I believe this is the error #60 suggested a fix for. In my case the edit suggested resolved the issue

nicolai-wachenschwan, 2023-12-18:
sorry it took so long, is fixed now!

closed 2023-12-18 by nicolai-wachenschwan