nicolai-wachenschwan/arc-overhang-prusaslicer-integration #66: Turning off the Hilbert Curve infill and random lines of filament.

tadeastehan, 2024-01-20:
Hello, I want to ask if there is any way to turn off the Hilbert curve infill because my overhangs isn't that big for this infill. I just want to try arc overhangs. 😄 
PrusaSlicer version is 2.7.1 and I use MK4.

[image]

And I also want to ask if there is any change how to remove this "red" random infill:

[image]

Whole rewrited model:

[image]

SamuraiNinjaGuy, 2024-01-23:
I'm fairly sure you can disable the hilbert curve by changing "specialCoolingZdist" to 0. Be aware that you may get warping though, as warping mitigation is the point of the special cooling layer.

See #56 for more information about the random lines of filament (basically a known issue).

Wasupmacuz, 2024-04-12:
> I'm fairly sure you can disable the hilbert curve by changing "specialCoolingZdist" to 0. Be aware that you may get warping though, as warping mitigation is the point of the special cooling layer.

Unfortunately, setting `specialCoolingZdist=0` still places a single cooling layer. Some simple logic needs to be added to disable hilbert curves.

jperry004, 2024-04-15:
I'm having a lot of issues with the special cooling layers, so I'm disabling it by Setting specialCoolingZdist to a negative value. Thanks to Wasupmacuz for pointing this out! Perhaps this is because I'm using a 0.6 mm nozzle? 

Left is with special cooling layers, right is without them (and 10 minutes faster). Bottom is about the same, as expected. 

[image]

[image]