nicolai-wachenschwan/arc-overhang-prusaslicer-integration #73: no matching overhangs found->no arcs generated.

Hairyloon, 2024-02-24:
I'm sorry if I've missed a clear instruction, but I can't find it, and I just get the error: 
"no matching overhangs found->no arcs generated."
It goes on to say " If unexpected: look if restricting settings like 'minArea' or 'MinBridgeLength' are correct".  
I can find these terms in the script, but nowhere in Prusa, so what would be correct settings for these?

Wasupmacuz, 2024-04-09:
If your overhangs are small, the default script will not replace them. Decreasing `minArea` will decrease the size that your overhangs have to be in order to be replaced. Likewise, decreasing `MinBridgeLength` will look for smaller bridging sections to replace with arcs.

Hairyloon, 2024-05-02:
Where are these parameters?  (minArea and MinBridgeLength)

Wasupmacuz, 2024-05-05:
They’re in the `prusa_slicer_post_processing_script.py` file. Just past the script header and imports, there’s the function that creates the settings dictionary. In the dictionary, the parameters will look like `”MinArea”:` and `”MinBridgeLength”:`, followed by their assigned values. Lower these values and the script will look for smaller overhangs 👍

Hairyloon, 2024-05-06:
Great, thank you. I'll try that.