nicolai-wachenschwan/arc-overhang-prusaslicer-integration #74: The code doesn't work with orca produced gcode

vgdh, 2024-03-21:
It throw an error about can't find a nozzle diameter settings. 
Orca generate gcode different somehow from prusa slicer.

May it possible to make the code to accept parameters or make some default values for the settings in order to prevent exceptions if settings is not defined in the gcode file?

jperry004, 2024-04-15:
Upload the gcode file and I'll look at it.

vgdh, 2024-04-15:
@Rickthebig is the author of the issue and he append the project below to the original issue.
https://github.com/SoftFever/OrcaSlicer/files/14665397/Carl_the_Cubeish_Test_Joint.zip

vgdh, 2024-04-15:
@Rickthebig please add the gcode if the problem is still there.

Rickthebig, 2024-04-15:
> @Rickthebig please add the gcode if the problem is still there.

What do you mean? Should I retry the postprocessor script? Where is the improved version?
I postponed printing this particular model until a viable solution is available. So the problem is still there, so to say.

jperry004, 2024-04-16:
> > @Rickthebig please add the gcode if the problem is still there.
> 
> What do you mean? Should I retry the postprocessor script? Where is the improved version? I postponed printing this particular model until a viable solution is available. So the problem is still there, so to say.

Upload a gcode file and we'll see if it's possible/easy to add support for Orca.

Rickthebig, 2024-04-16:
OK, I am uploading the corresponding gcode file (renamed).
[Carl_the_Cubeish_Test_Joint__PLA.txt](https://github.com/nicolai-wachenschwan/arc-overhang-prusaslicer-integration/files/15003655/Carl_the_Cubeish_Test_Joint__PLA.txt)
I will stay tuned since I am very interested to get this arc overhang working, even if it is not a general option to be used blindly.

jperry004, 2024-04-17:
> @jperry004 OK, I am uploading the corresponding gcode file (renamed). [Carl_the_Cubeish_Test_Joint__PLA.txt](https://github.com/nicolai-wachenschwan/arc-overhang-prusaslicer-integration/files/15003655/Carl_the_Cubeish_Test_Joint__PLA.txt) I will stay tuned since I am very interested to get this arc overhang working, even if it is not a general option to be used blindly.

Nice, thanks for uploading. I did get it to run, and it's fairly easy - there's just a few things that are named differently in Orca. However no arcs were generated. It looks like most of the overhangs in this model are very small, too small for arcs. The default minimum is 50mm^2. However I found 1 overhang, at around line 120120, which is large enough but still was not generated. I'm not sure why that is, and more advanced knowledge than mine is needed from there. If you want to continue it would be best to start with some simple geometry object. Perhaps the easiest things for now is to use Prusa slicer. 

[image]

[image]

jperry004, 2024-04-17:
> > > @jperry004 OK, I am uploading the corresponding gcode file (renamed). [Carl_the_Cubeish_Test_Joint__PLA.txt](https://github.com/nicolai-wachenschwan/arc-overhang-prusaslicer-integration/files/15003655/Carl_the_Cubeish_Test_Joint__PLA.txt) I will stay tuned since I am very interested to get this arc overhang working, even if it is not a general option to be used blindly.
> > 
> > 
> > Nice, thanks for uploading. I did get it to run, and it's fairly easy - there's just a few things that are named differently in Orca. However no arcs were generated. It looks like most of the overhangs in this model are very small, too small for arcs. The default minimum is 50mm^2. However I found 1 overhang, at around line 120120, which is large enough but still was not generated. I'm not sure why that is, and more advanced knowledge than mine is needed from there. If you want to continue it would be best to start with some simple geometry object. I tried to open the STL in Prusa, but it wouldn't open for me. Perhaps the easiest things for now is to use Prusa slicer.
> > ![image](https://private-user-images.githubusercontent.com/128223478/323108222-c23970f8-9557-452c-88d4-3a8e9a19d9c6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMzNTU0MDAsIm5iZiI6MTcxMzM1NTEwMCwicGF0aCI6Ii8xMjgyMjM0NzgvMzIzMTA4MjIyLWMyMzk3MGY4LTk1NTctNDUyYy04OGQ0LTNhOGU5YTE5ZDljNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQxN1QxMTU4MjBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01OTVlNzJiODcxNGQzNTQxYWIwYmRmYTdiY2JiMTgxN2EyM2I0NGViMjM0MjIxMGM3MmJmNDQ0YTYyODY5YWRiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.5Xl92aOsLGRuUGm2UTaPrrsCAgQkyewQtDRKsY-lEkI)
> > ![image](https://private-user-images.githubusercontent.com/128223478/323108497-478b52ea-d68e-41bd-b0d2-976fa9860172.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMzNTU0MDAsIm5iZiI6MTcxMzM1NTEwMCwicGF0aCI6Ii8xMjgyMjM0NzgvMzIzMTA4NDk3LTQ3OGI1MmVhLWQ2OGUtNDFiZC1iMGQyLTk3NmZhOTg2MDE3Mi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQxN1QxMTU4MjBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mM2YzODc1ZjlkM2RjYjEwNmRmMDk0ZmE4NWRmMWQwYTU5NGUwNjFjYTU2OTU4ZDJkZDNmZTVkMjU3MjEyMzgwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.c0yavZDZ8eynKWvEyqxL_2U_q2HQZTO31QRc3gQaVrg)
> 
> Yes, the overhang you detected at 120120 is the one I am coping with. But in fact, it begins at 120156 (a bit later). The overhang mark seems to be a bit displaced.
> 
> Here is the crucial layer where overhang comes to play and showing a pronounced overhang:
> 
> ![image](https://private-user-images.githubusercontent.com/151161212/323210453-f67739a9-b883-4465-b189-e1f287036442.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMzNjM5NTgsIm5iZiI6MTcxMzM2MzY1OCwicGF0aCI6Ii8xNTExNjEyMTIvMzIzMjEwNDUzLWY2NzczOWE5LWI4ODMtNDQ2NS1iMTg5LWUxZjI4NzAzNjQ0Mi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQxN1QxNDIwNThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kMTRmMGQ5ZWQzMDA3Zjc3NzJkYTQzZGVhZTFlNTliM2Y4ODQ4NDhmOTg2YjhmYTc3YmQyZjhjNmIwMDY0NTM4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.6PlCCPgsf06IeH2NcgtlhaTRtajZI0ZKbTdGdiW88fU)
> 
> Trying to build this layer in midair with conventional overhang technique will definitely fail.
> 
> Prior to building this layer, you can find the overhang mark at line 120125:
> 
> ![image](https://private-user-images.githubusercontent.com/151161212/323216720-858789f8-627a-4a4f-b308-b339c0f9bf35.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMzNjM5NTgsIm5iZiI6MTcxMzM2MzY1OCwicGF0aCI6Ii8xNTExNjEyMTIvMzIzMjE2NzIwLTg1ODc4OWY4LTYyN2EtNGE0Zi1iMzA4LWIzMzljMGY5YmYzNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQxN1QxNDIwNThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kMmM3YjhjMTE3ZjU1ZTVmN2Q5ODEwNDUyNWZjNGRiNzY4NzNhNTY5ZDk5ZmNiZjdkNWI3YjJmYTNlOTJhMTU0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.0AUWkc_az2OcK93fayjAkWyhJsCnMo02-aM4xfIrDik)
> 
> which is a bit too early in my understanding. The actual overhang begins a couple of moves later.
> 
> See also the mark "120791;TYPE:Overhang wall".
> 
> ![image](https://private-user-images.githubusercontent.com/151161212/323214369-fdd6be61-3773-474f-ab1a-38ad01102178.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMzNjM5NTgsIm5iZiI6MTcxMzM2MzY1OCwicGF0aCI6Ii8xNTExNjEyMTIvMzIzMjE0MzY5LWZkZDZiZTYxLTM3NzMtNDc0Zi1hYjFhLTM4YWQwMTEwMjE3OC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzR [... cut ...]
Thanks for the screenshots, really helpful! That is one overhang definitely large enough and should be generated. I tried the englarged gcode, but same problem. I suspect the issue is a bit deeper in how polygons are being detected. As if the lines are being drawn differently in Orca. 

I created a script that will modify Orca gcode files to work with the arc overhang script. I thought this would be easiest, as it doesn't require modifying the actual arc overhang script. It's located here: 

https://github.com/jperry004/arc-overhang-prusaslicer-integration/tree/orca-tests-issue-78

that should be public I believe. To use it: 

[image]

Rickthebig, 2024-04-17:
> The layers and polys printouts are debug only

"Layer 96: 14 overhangs found"

That is the point where the script should kick in.
I downloaded the conversion script and will give it a try today later.
I think the problem could be that the original overhang has a very particular form and is filled with arcs due to the round perimeter - substituting arcs with arcs...

So to begin I will try more normal cases and see if it works at all.

I will report back - thanks sofar.

jperry004, 2024-04-17:
One cool thing I noticed is after using the converter script you can then look at the gcode with Prusa gcode visualizer. So I made a new STL with the same rough shape and it's able to generate the arcs no problem, first try. Behold, fake Carl: 

[image]

I still can't quite figure out what is stopping the Orca file from working though.

caxtell, 2024-04-21:
I have experienced this as well when using orca slicer and have tried using the convert_orca_prusa-script with little success. I was able to capture from the terminal a clue. 
 Type Error: unsupported operand type(s) for *: 'float' and 'Nonetype'
[process exited with code 1 (0x00000001)]

jperry004, 2024-04-21:
> I have experienced this as well when using orca slicer and have tried using the convert_orca_prusa-script with little success. I was able to capture from the terminal a clue. Type Error: unsupported operand type(s) for *: 'float' and 'Nonetype' [process exited with code 1 (0x00000001)]

Could you upload the gcode file that you're trying to convert?

caxtell, 2024-04-21:
Thanks in advance for the advise. 

I tried also setting up the script in Prusa Slicer and configured the printer settings to work with my bambu P1S, just for clarification, and I got the exact same error when trying to Export to G-code. I believe it maybe something inherent to the printer custom g-code settings.

(Note: Had to split the file in two to upload it du
[7684-conti.gcode.txt.txt](https://github.com/nicolai-wachenschwan/arc-overhang-prusaslicer-integration/files/15052751/7684-conti.gcode.txt.txt)
e to size)
[Uploading 7684.gcode.txt…]()

boromyr, 2024-08-24:
Has anyone managed to get it to work in orcaslicer? I am getting this error.
```py
Traceback (most recent call last):
  File "C:\Users\PC\AppData\Roaming\OrcaSlicer\arc.py", line 1197, in <module>
    main(gCodeFileStream,path2GCode, skipInput)
  File "C:\Users\PC\AppData\Roaming\OrcaSlicer\arc.py", line 106, in main
    parameters=makeFullSettingDict(gCodeSettingDict)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\PC\AppData\Roaming\OrcaSlicer\arc.py", line 53, in makeFullSettingDict
    "ExtendIntoPerimeter":1.5*gCodeSettingDict.get("perimeter_extrusion_width"), #min=0.5extrusionwidth!, extends the Area for arc generation, put higher to go through small passages. Unit:mm
                          ~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
TypeError: unsupported operand type(s) for *: 'float' and 'NoneType'
```

nicolai-wachenschwan, 2024-12-20:
The Problem with orca are some differences in the gcode Generation and the naming of the config. The Script only works with prusa slicer :/