nicolai-wachenschwan/arc-overhang-prusaslicer-integration #77: Arcs not generated after removing bridge infill

Wasupmacuz, 2024-04-11:
The post-processing script outputs this:
```
Layer 201: No ExternalPerimeterPolys found in prev Layer
Skipping Polygon because no StartLine Found
```
and changed this bridging:
[image]
into a hole like this:
[image]
This is happening because the external perimeter is a line and doesn't contain enough points to make a polygon. The `makePolygonFromGCode` function returns `None`, due to this. Following this, the `makeExternalPerimeter2Polys` function does not update the `self.extPerimeterPolys` list, causing the `makeStartLineString` function to return `None, None`. When this occurs, arc generation skips the current polygon, leaving a hole. I haven't looked into a fix yet.

nicolai-wachenschwan, 2024-04-12:
Thank you for the Report! , finding a starting point is a tricky problem with the edge cases…

Wasupmacuz, 2024-04-13:
@nicolai-wachenschwan My initial speculation was incorrect. The bug actually had to do with the logic in the `makeGeometryFromGCode` function. Would you like me to submit another pull request that fixes this issue *as well* as #56? Or should I wait until you've had the chance to look over my first pull request for #56?
[image]

jperry004, 2024-04-14:
I tested this fix, it works for me. Thanks for posting!