nicolai-wachenschwan/arc-overhang-prusaslicer-integration #84: Script error when generating g-code

plandevida, 2024-10-31:
Hi,

I am having this issue where the script will fail reporting the module shapely is not found:

[image]

I have installed all dependencies as explained in the documentation:

[image]

I have install python3 with brew on macOS.

plandevida, 2024-12-19:
Hi!

I got an email with your @nicolai-wachenschwan response but I don't see your comment here anymore.

> I'm trying to wrap my head around what could be the issue here. A quick glance shows that everything is in order (though I'm definitely no expert). Try running the script through the terminal instead of using Orca.

I got the same error on the terminal, but I managed to execute it in the end getting this another error:

```
$ python3 prusa_slicer_post_processing_script.py Cube_0.2mm_PLA_MINI.gcode
Traceback (most recent call last):
  File "/Users/dserrano/Downloads/prusa_slicer_post_processing_script.py", line 1197, in <module>
    main(gCodeFileStream,path2GCode, skipInput)
    ~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/dserrano/Downloads/prusa_slicer_post_processing_script.py", line 106, in main
    parameters=makeFullSettingDict(gCodeSettingDict)
  File "/Users/dserrano/Downloads/prusa_slicer_post_processing_script.py", line 53, in makeFullSettingDict
    "ExtendIntoPerimeter":1.5*gCodeSettingDict.get("perimeter_extrusion_width"), #min=0.5extrusionwidth!, extends the Area for arc generation, put higher to go through small passages. Unit:mm
                          ~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
TypeError: unsupported operand type(s) for *: 'float' and 'NoneType'
```

nicolai-wachenschwan, 2024-12-20:
The First issue was, that the Python you are using for the script did not have shapely installed. The Second error is, that my script only works with Prusa sclicer. There might be a fork where someone adapted it for orca :)

closed 2024-12-20 by nicolai-wachenschwan

plandevida, 2024-12-20:
okay the first issue I worked around activating the venv first and executing the script.

I see, is it very different to make it work with Orca?